raulmur/ORB_SLAM #108: Building on Raspberry PI

Clapton-Is-God, 2016-05-04:
Is it possible for ORB Slam to work on raspberry pi? I have been trying to build it on a rpi 3 with raspbian jessie and ROS Indigo. I get the following error when trying to build g2o using make:

_Scanning dependencies of target g2o
[  3%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o
*** Error in `/usr/bin/c++': double free or corruption (top): 0x00c7a348 ***
CMakeFiles/g2o.dir/build.make:54: recipe for target 'CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o' failed
make[2]: *** [CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o] Aborted
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/g2o.dir/all' failed
make[1]: *** [CMakeFiles/g2o.dir/all] Error 2
Makefile:76: recipe for target 'all' failed
make: *** [all] Error 2_

Does anyone know how to solve this? Any help would be appreciated. Regards.

riematrix, 2016-06-28:
@Clapton-Is-God  Hi, the key message is "double free or corruption". I used to compile the project on Pi3 after adding an option in CMakelists.txt but i can't remember it now. However i couldn't run the program on my raspberry pi. It keeps showing thread error or bus error that I almost give up.

fannief, 2016-06-30:
I found the solution to run ORB_SLAM2 on the RPi2 but it is still 3x slower than the normal performance: 

For the compilation: 
- Enable SWAP in the dphys-swapfile to 1536MB (enough for me on the RPi2)
- Add -j2 after all the make instructions in the build.sh (that's the trick!)
- Use certain flags on your CMakeLists.txt : -Wall -mfpu=neon -O3 -pthread -std=c++11and enable neon with add_definitions("-DENABLE_NEON"). Don't use the -march=armv7-a flag, you may get a virtual function error (thread).  
If you try to compile the ROS version you might get an issue with Opencv, it is because you have two versions of opencv installed on your RPi, you might have to change some opencv headers on the code. I found that the version without ROS works better on the RPi. 

For the ORB Vocabulary (too big for the RPi): 
The best to do would be to create a new ORB vocabulary with DBoW2 but I simply deleted 4/5 of the ORBvoc.txt and I still get good results. 

The visualization is very slow on the RPi, then I used ssh to get the visualization on another computer.

riematrix, 2016-07-01:
Thank you so much for sharing this! @fannief 

BTW, the lode process could be much faster if you merge this [binary vocabulary version](https://github.com/raulmur/ORB_SLAM2/pull/21). On my Pi3 it takes about 10 seconds.

hyqneuron, 2016-11-04:
@fannief May I ask what kind of framerate are you getting and what settings are you using? I used the default compilation flags and tried the default setting (TUM1.yaml) and seemed to be getting 2~3fps on RPi2 with a webcam.

hyqneuron, 2016-11-04:
@riematrix How's the performance on RPi3?

fannief, 2016-11-04:
The settings depend on the dataset you are using, I tried to decrease at maximum the number of ORB features and the number of pyramid scales (most time consuming). I did also some modifications inside the code to get a framerate around 3-4fps for the KITTI dataset (07) and ~3fps for real time dataset with stereo camera, VGA images, 20fps. You can also try to add NEON instructions to speed up a bit more (I didn't try, but why not). 
I couldn't record images and run ORB SLAM at the same time on the RPi, I had to use WiFi/Ethernet to get the frames in real time. RPi2 is very limited :/ I didn't try with RPi3

hyqneuron, 2016-11-05:
@fannief 20fps with VGA images?! May I know what exact settings are you using (number of points, scale and number of levels)? I'm using a VGA webcam and I'm only getting about 3fps on RPi3, with nFeatures=500, scaleFactor=1.5 and nLevels=3.

fannief, 2016-11-05:
20fps is just the input, my stereo camera frame rate. ~3fps is the output. It will be hard to do better than 4fps.

diophantine86, 2017-04-18:
i build orbslam2 on raspberry pi3. but this occurs eigen dense storage error.

zanazakaryaie, 2017-09-28:
Hello

Thanks for sharing

I have installed it on Rpi2 successuflly but when I run kitti example it first loads vocabulary and then throws an error:

Pangolin X11: Unable to retrieve framebuffer options

What should I do?

fannief, 2017-09-30:
https://github.com/raulmur/ORB_SLAM2/issues/4 
@'sunstarchan commented on Feb 1, 2016'