raulmur/ORB_SLAM #109: Error during build, rospack

gunshi, 2016-05-06:
I am trying to compile ORB_SLAM on OSX 10.11.12 with ROS Indigo. i have opencv 3.1 installed under /usr/local/lib and /usr/local/include (default), and also opencv2.4 under /usr/local/Cellar(through brew), which I can link to, in case opencv 3 is not compatible.

I successfully built g2o and DBoW2 and then on running cmake .. -DROS_BUILD_TYPE=Release, (after appending to ROS_PACKAGE_PATH), I get

> -- Found PythonInterp: /usr/local/bin/python (found version "2.7.11") 
> [rosbuild] Building package ORB_SLAM
> Failed to invoke /Users/gunshigupta/ros_catkin_ws/install_isolated/bin/rospack dept-manifests ORB_SLAM
> Traceback (most recent call last):
>   File "/usr/local/lib/python2.7/site-packages/rosdep2/__init__.py", line 45, in <module>
>     from .lookup import RosdepDefinition, RosdepView, RosdepLookup, \
>   File "/usr/local/lib/python2.7/site-packages/rosdep2/lookup.py", line 44, in <module>
>     from .sources_list import SourcesListLoader
>   File "/usr/local/lib/python2.7/site-packages/rosdep2/sources_list.py", line 34, in <module>
>     import tempfile
>   File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/tempfile.py", line 32, in <module>
>     import io as _io
>   File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/io.py", line 51, in <module>
>     import _io
> ImportError: dlopen(/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_io.so, 2): Symbol not found: __PyCodecInfo_GetIncrementalDecoder
>   Referenced from: /usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_io.so
>   Expected in: flat namespace
>  in /usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload/_io.so
> [rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?
> 
> 
> CMake Error at /Users/gunshigupta/ros_catkin_ws/install_isolated/share/ros/core/rosbuild/public.cmake:129 (message):
>   
> 
>   Failed to invoke rospack to get compile flags for package 'ORB_SLAM'.  Look
>   above for errors from rospack itself.  Aborting.  Please fix the broken
>   dependency!
> 
> Call Stack (most recent call first):
>   /Users/gunshigupta/ros_catkin_ws/install_isolated/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
>   CMakeLists.txt:4 (rosbuild_init)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/Users/gunshigupta/Desktop/4sfm/ORB_SLAM/build/CMakeFiles/CMakeOutput.log".
> 

I also removed the line   <depend package="opencv"/> from manifest.xml, but got the same error.

How can i fix it? I have rosdep 0.11.4.
Thanks.

pbadger, 2016-05-31:
I had this problem and it was due to some overlapping python versions (see this: http://answers.ros.org/question/53110/rviz-runtime-error-in-groovy-at-mac-os-xhomebrew/) 

For me, the solution was to brew uninstall --force python

archenroot, 2017-06-04:
Yes, related to versions of python, on Gentoo usually you have multiple versions, I was lucky unfortunately only with 2.7, 3.4 didn't work. But as per my understanding ROS 1.* supports only 2.7 officially.

archenroot, 2017-06-05:
Well, I removed the stacktrace issues reported by Python itself, but now facing the second issue (same as you):
```
zangetsu@ares ~/proj/neural-networks/ORB_SLAM2 $ ./build_ros.sh 
Building ROS nodes
mkdir: adresář „build“ nelze vytvořit: Soubor již existuje
[rosbuild] Building package ORB_SLAM2
Failed to invoke /usr/bin/rospack deps-manifests ORB_SLAM2
[rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?

CMake Error at /usr/share/ros/core/rosbuild/public.cmake:129 (message):
  

  Failed to invoke rospack to get compile flags for package 'ORB_SLAM2'.
  Look above for errors from rospack itself.  Aborting.  Please fix the
  broken dependency!

Call Stack (most recent call first):
  /usr/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/zangetsu/proj/neural-networks/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** Nezadány žádné cíle a žádné makefily nenalezeny. Konec.

```