raulmur/ORB_SLAM #110: opencv error

xxsong5, 2016-05-11:
Start processing sequence ...
Images in the sequence: 1893

New map created with 563 points
OpenCV Error: Assertion failed (m.dims >= 2) in Mat, file /home/xxsong5/Downloads/opencv-2.4.12/modules/core/src/matrix.cpp, line 269
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/xxsong5/Downloads/opencv-2.4.12/modules/core/src/matrix.cpp:269: error: (-215) m.dims >= 2 in function Mat

closed 2016-05-11 by xxsong5

hustchenli, 2020-07-31:
@xxsong5 I got similar error. Did you solve it?

pmbarile, 2020-07-31:
No I did not! Sorry

hustchenli, 2020-07-31:
@xxsong5 I solved it by connecting camera with another USB interface. But the program can only run for a few seconds and crashed again with the same error. I'm trying to figure it out.