raulmur/ORB_SLAM #114: Not an issue but help needed during ORB SLAM build

samsungecos, 2016-06-15:
Hello, 
I got the following error messages.  Did I miss some library or I incorrectly installed dependencies?

hiteshc@2939993978-LAB5:~/orb/ORB_SLAM/build$ make
Scanning dependencies of target rospack_genmsg_libexe
[  0%] Built target rospack_genmsg_libexe
Scanning dependencies of target rosbuild_precompile
[  0%] Built target rosbuild_precompile
Scanning dependencies of target ORB_SLAM
[  5%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/main.cc.o
[ 11%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o
[ 16%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/LocalMapping.cc.o
[ 22%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/LoopClosing.cc.o
[ 27%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/ORBextractor.cc.o
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::ComputeKeyPoints(std::vector<std::vector<cv::KeyPoint> >&)’:
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc:607:63: error: ‘FAST’ was not declared in this scope
                 FAST(cellImage,cellKeyPoints[i][j],fastTh,true);
                                                               ^
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc:616:34: error: ‘ORB’ has not been declared
                 if( scoreType == ORB::HARRIS_SCORE )
                                  ^
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc:683:17: error: ‘KeyPointsFilter’ has not been declared
                 KeyPointsFilter::retainBest(keysCell,nToRetain[i][j]);
                 ^
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc:699:13: error: ‘KeyPointsFilter’ has not been declared
             KeyPointsFilter::retainBest(keypoints,nDesiredFeatures);
             ^
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::operator()(cv::InputArray, cv::InputArray, std::vector<cv::KeyPoint>&, cv::OutputArray)’:
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc:760:82: error: ‘GaussianBlur’ was not declared in this scope
         GaussianBlur(workingMat, workingMat, Size(7, 7), 2, 2, BORDER_REFLECT_101);
                                                                                  ^
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::ComputePyramid(cv::Mat, cv::Mat)’:
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc:800:78: error: ‘INTER_LINEAR’ was not declared in this scope
             resize(mvImagePyramid[level-1], mvImagePyramid[level], sz, 0, 0, INTER_LINEAR);
                                                                              ^
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc:800:90: error: ‘resize’ was not declared in this scope
             resize(mvImagePyramid[level-1], mvImagePyramid[level], sz, 0, 0, INTER_LINEAR);
                                                                                          ^
/home/hiteshc/orb/ORB_SLAM/src/ORBextractor.cc:803:80: error: ‘INTER_NEAREST’ was not declared in this scope
                 resize(mvMaskPyramid[level-1], mvMaskPyramid[level], sz, 0, 0, INTER_NEAREST);
                                                                                ^
make[2]: *** [CMakeFiles/ORB_SLAM.dir/src/ORBextractor.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
make: *** [all] Error 2

ram651991, 2016-07-01:
Faced the same issue. Was able to solve it using @aaronma37 comment in https://github.com/raulmur/ORB_SLAM/issues/44

tahmine-iaun, 2016-09-05:
Hello. I faced with this err too but iam using ros hydro! Would you please help me?

ram651991, 2016-09-08:
@tahmine-iaun ... ROS hydro shouldn't be a problem. Did you try adding header files as indicated by @aaronma37's comment in #44 ??

tahmine-iaun, 2016-09-10:
@ram651991  tnx for your reply.. you konw things become a little complicated to me... just take a look at this message while iam trying to implement orbslam! 
[image]

You know the ver of Eigen which is used here" sudo apt-get install libeigen3-dev" is under exception .i had to install it's new ver manually.
could you pls help me with that? i've searched the net but found nothing useful

roshdii, 2018-03-11:
this issue can help you https://github.com/raulmur/ORB_SLAM/issues/131