raulmur/ORB_SLAM #119: ORB_SLAM installation on Ubuntu 16.04

romdos, 2016-07-27:
Is it possible to install ORB_SLAM/ORB_SLAM2 on last version of Ubuntu (Xenial 16.04) without black magic? I know that the recommendation is to use Ubuntu 14.04 according to Github, but I currently have last version and I don't really want to change it or install 14 together with 16.

Now the problem is like in https://github.com/raulmur/ORB_SLAM2/issues/68#issuecomment-223621426

romdos, 2016-07-29:
The problem partially solved thanks to #https://github.com/raulmur/ORB_SLAM2/issues/68#issuecomment-220739225

chaiein, 2017-01-13:
I am using  ORB_SLAM2 in ubuntu 16. So it works.

romdos, 2017-01-13:
Oh, yes. I've figured out this. What camera calibration method did you use?

chaiein, 2017-01-13:
using this http://wiki.ros.org/camera_calibration  I got the parameters and updated in the yaml file.

wangzhaoguang, 2017-09-13:
[image]

Why this ?

roseraven, 2018-06-27:
i met the same problem, have you solved the problem? @wangzhaoguang