raulmur/ORB_SLAM #124: problem when building orbslam on ubuntu 14.04 (vmware)

ietr, 2016-11-09:
Hello guys, 

I got a classic problem but cannot solve it with classic solution.

```
arifle@ubuntu:~/orbslam/ORB_SLAM/build$ sudo cmake ..
[sudo] password for arifle: 
CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /core/rosbuild/rosbuild.cmake

CMake Error at CMakeLists.txt:5 (rosbuild_init):
  Unknown CMake command "rosbuild_init".

-- Configuring incomplete, errors occurred!
See also "/home/arifle/orbslam/ORB_SLAM/CMakeFiles/CMakeOutput.log".
arifle@ubuntu:~/orbslam/ORB_SLAM/build$ source /opt/ros/indigo/setup.bash 
arifle@ubuntu:~/orbslam/ORB_SLAM/build$ echo $ROS_PACKAGE_PATH
/opt/ros/indigo/share:/opt/ros/indigo/stacks
arifle@ubuntu:~/orbslam/ORB_SLAM/build$ export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/arifle/orbslam/ORB_SLAM
arifle@ubuntu:~/orbslam/ORB_SLAM/build$ 
arifle@ubuntu:~/orbslam/ORB_SLAM/build$ sudo cmake ..
CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /core/rosbuild/rosbuild.cmake

CMake Error at CMakeLists.txt:5 (rosbuild_init):
  Unknown CMake command "rosbuild_init".

-- Configuring incomplete, errors occurred!
See also "/home/arifle/orbslam/ORB_SLAM/CMakeFiles/CMakeOutput.log".
arifle@ubuntu:~/orbslam/ORB_SLAM/build$ 

```

See, nothing changed after setting the path.

chaiein, 2016-11-10:
Try this 
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/arifle/orbslam/ORB_SLAM/
In this slash is changed.

saurabhagrawal3, 2016-11-25:
Hi @chaiein,

I too have to same error when trying to cmake in build dir.

tried your suggestion too but didn't work. got the same error.
 Any other suggestion please do comment. Thanks

Hi @ietr,

Please post the solution if you have found one. 

Thanks

arifle, 2016-12-07:
Not yet, and you?  @saurabhagrawal3

arifle, 2016-12-07:
Not yet. @saurabhagrawal3

chaiein, 2017-01-12:
Are you using the qtcreator for this? If so , first give 
$source /opt/ros/<ros distribution >/setup.bash , then start the qtcreator

ckchng, 2019-11-04:
> Hi guys, anyone found any solution on this? @arifle @saurabhagrawal3

Got it, @chaiein's solution worked for me.