raulmur/ORB_SLAM #126: error: Failed to open settings file

test-ing, 2016-12-30:
Hi,

I am trying to run ORB SLAM 2 with ROS on the examples but when I do this:
$ rosrun ORB_SLAM2 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml true
I get the following error:
Input sensor was set to: Stereo
Failed to open settings file at: Examples/Stereo/EuRoC.yaml

However when I tried the run ORB SLAM 2 without ROS on the examples, it worked so I guess the issue is not with the settings file.

laxnpander, 2017-01-24:
I can confirm this. Did you find a solution?

idolezel, 2017-07-20:
Resolved here:
https://github.com/raulmur/ORB_SLAM2/issues/227

ashokyendluri, 2018-03-16:
I'm using ORB SLAM2 to process images from a ZED Stereo camera in real time and I ran into this issue 
Failed to open settings file at: /home/Desktop/zed-opencv/zed_ygx.yaml. I've tried both relative and absolute paths. There's only one version of OPENCV (3.4.0) installed.

Also, I'm succesfully able to compile and run the code samples in the Examples folder but the code doesn't work with my stereo camera. 
Any help is appreciated!
Thanks in advance

laxnpander, 2018-03-18:
@ashok-y  Any updates? Can't really think of a reason other than that the file does not exist.

ashok-y, 2018-03-19:
No, I've recompiled the ORB SLAM and OpenCV again but the issue still exists.