raulmur/ORB_SLAM #136: How to use ORB_SLAM with ROS Navigation Stack?

dustedDuke, 2017-04-15:
I am currently working on a mobile robot project, in which I want to implement ROS Navigation to be able to set goal pose and avoid obstacles. The main problem is, that I don't have a Kinect or a laser sensor, so I'm not able to send any PointCloud or LaserScan messages. I also have MPU 6050, so I can get some odometry data. For now I can't even save a map, that is created by ORB_SLAM in any way. I tried to relay ORB_SLAM/Map topic to /map to use it for saving via map_server map_saver, but map_server couldn't find it. Is there any way to use ORB_SLAM instead of PointCloud/LaserScan? If it's not possible, how can I use it for navigation and sending goals?

DalderupMaurice, 2017-05-10:
I have the same problem, did you find any solution yet? @dustedDuke

astronaut71, 2018-06-04:
I have also the same problem. Any solutions yet?

pmbarile, 2018-06-06:
no it's been well over a year and nobody responded with a solution.