raulmur/ORB_SLAM #138: [rospack] Error: package 'orb_slam' not found

OkanYuksel, 2017-05-07:
Hi everyone, 
Please help me..

I added this line in bashrc
export ROS_PACKAGE_PATH=/opt/ros/indigo/share:/opt/ros/indigo/stacks:/home/turtlebot/orbslam

but I take this error when execute, "rosrun orbslam orbslam /home/turtlebot/orbslam/trunk/Data/ORBvoc.yml /home/turtlebot/orbslam/trunk/Data/Settings.yaml"

Thank you so much.

NehaMazhar, 2022-05-21:
Hi, i am facing same error. have changed the path as to following
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/syed/Dev/ORB_SLAM3/Examples/ROS

i cant find file home/syed/Dev/ORB_SLAM3/Examples/ROS so I hade a new file as ROS. still didnt resolve issue. PLease help. thanks

Lynx1129, 2023-03-23:
did you resolve it?