raulmur/ORB_SLAM #141: But the last step has issues,

yanyan-li, 2017-06-19:
It's almost done. But the last step has issues now. Can somebody help?  I have no ideas about it. Thanks.
---------------------------------------------
[ 84%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/KeyFrameDatabase.cc.o
[ 89%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Sim3Solver.cc.o
[ 94%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Initializer.cc.o
[100%] Linking CXX executable ../bin/ORB_SLAM
CMakeFiles/ORB_SLAM.dir/src/main.cc.o: In function `main':
main.cc:(.text.startup+0xbc8): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
main.cc:(.text.startup+0xccb): undefined reference to `ros::package::getPath(std::string const&)'
main.cc:(.text.startup+0xe5c): undefined reference to `ros::package::getPath(std::string const&)'
main.cc:(.text.startup+0x13b4): undefined reference to `ros::package::getPath(std::string const&)'
main.cc:(.text.startup+0x193b): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o: In function `ORB_SLAM::Tracking::CreateInitialMap(cv::Mat&, cv::Mat&)':
Tracking.cc:(.text+0x21e4): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
Tracking.cc:(.text+0x229f): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o: In function `ORB_SLAM::Tracking::Run()':
Tracking.cc:(.text+0x2946): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o: In function `ORB_SLAM::Tracking::GrabImage(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)':
Tracking.cc:(.text+0x4589): undefined reference to `cv_bridge::toCvShare(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, std::string const&)'
Tracking.cc:(.text+0x5e11): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o: In function `ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, void>::deserialize(ros::SubscriptionCallbackHelperDeserializeParams const&)':
Tracking.cc:(.text._ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE[_ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE]+0x604): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/ORB_SLAM.dir/src/LoopClosing.cc.o: In function `ORB_SLAM::LoopClosing::CorrectLoop()':
LoopClosing.cc:(.text+0x2a47): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/ORB_SLAM.dir/src/FramePublisher.cc.o: In function `ORB_SLAM::FramePublisher::FramePublisher()':
FramePublisher.cc:(.text+0x1c3a): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
CMakeFiles/ORB_SLAM.dir/src/MapPublisher.cc.o: In function `ORB_SLAM::MapPublisher::MapPublisher(ORB_SLAM::Map*)':
MapPublisher.cc:(.text+0xb7e): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/ORB_SLAM.dir/build.make:1645: recipe for target '../bin/ORB_SLAM' failed
make[2]: *** [../bin/ORB_SLAM] Error 1
CMakeFiles/Makefile2:324: recipe for target 'CMakeFiles/ORB_SLAM.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
-------------------------------------------------------
my OS is Ubuntu16.04. Thanks a lot.

wangzhaoguang, 2017-07-22:
Actually I got this too ! 
The last step for compiling ORB-SLAM. 
Any hints ? 
 cheers !! 

-------------

CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o：在函数‘ORB_SLAM::Tracking::GrabImage(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)’中：
Tracking.cc:(.text+0x494b)：对‘cv_bridge::toCvShare(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
CMakeFiles/ORB_SLAM.dir/src/FramePublisher.cc.o：在函数‘ORB_SLAM::FramePublisher::PublishFrame()’中：
FramePublisher.cc:(.text+0x24e5)：对‘cv_bridge::CvImage::toImageMsg() const’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/ORB_SLAM.dir/build.make:1567: recipe for target '../bin/ORB_SLAM' failed
make[2]: *** [../bin/ORB_SLAM] Error 1
CMakeFiles/Makefile2:324: recipe for target 'CMakeFiles/ORB_SLAM.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

---------------

wangzhaoguang, 2017-07-23:
hello world, 
for the building problem I encountered,  I got through via:
see the attached pictures. 

cheers.
[image]
[image]

jtbon20, 2017-08-01:
@wangzhaoguang Is there anything else you changed? I deleted Opencv2 as well and it still didn't work

wangzhaoguang, 2017-09-13:
No
i did not change other parts.

jinchenglee, 2017-11-08:
Check your gcc version. I met similar problem in ROS nodes building. Gcc 4.9 has linking problem, but gcc 5.4 works fine.

whyscience, 2017-12-28:
refer to this, solve it for ROS catkin build on Ubuntu 16.04, thanks.

Debbieliang9, 2021-04-19:
Where shall we place the <package> code? @wangzhaoguang Thanks!