raulmur/ORB_SLAM #149: Need help with build installation

jtbon20, 2017-08-01:
Trying to install with ROS Kinetic... not sure if this will work, during cmake on the whole package ran into the following errors:

/usr/Desktop/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::ComputeKeyPoints(std::vector<std::vector<cv::KeyPoint> >&)’:
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc:607:63: error: ‘FAST’ was not declared in this scope
                 FAST(cellImage,cellKeyPoints[i][j],fastTh,true);
                                                               ^
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc:616:34: error: ‘ORB’ has not been declared
                 if( scoreType == ORB::HARRIS_SCORE )
                                  ^
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc:683:17: error: ‘KeyPointsFilter’ has not been declared
                 KeyPointsFilter::retainBest(keysCell,nToRetain[i][j]);
                 ^
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc:699:13: error: ‘KeyPointsFilter’ has not been declared
             KeyPointsFilter::retainBest(keypoints,nDesiredFeatures);
             ^
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::operator()(cv::InputArray, cv::InputArray, std::vector<cv::KeyPoint>&, cv::OutputArray)’:
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc:760:82: error: ‘GaussianBlur’ was not declared in this scope
     GaussianBlur(workingMat, workingMat, Size(7, 7), 2, 2, BORDER_REFLECT_101);
                                                                              ^
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc: In member function ‘void ORB_SLAM::ORBextractor::ComputePyramid(cv::Mat, cv::Mat)’:
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc:800:78: error: ‘INTER_LINEAR’ was not declared in this scope
    resize(mvImagePyramid[level-1], mvImagePyramid[level], sz, 0, 0, INTER_LINEA
                                                                     ^
/usr/Desktop/ORB_SLAM/src/ORBextractor.cc:800:90: error: ‘resize’ was not declared in this scope
 resize(mvImagePyramid[level-1], mvImagePyramid[level], sz, 0, 0, INTER_LINEAR);
                                                                              ^
//usr/Desktop/ORB_SLAM/src/ORBextractor.cc:803:80: error: ‘INTER_NEAREST’ was not declared in this scope
      resize(mvMaskPyramid[level-1], mvMaskPyramid[level], sz, 0, 0, INTER_NEARE
                                                                     ^
CMakeFiles/ORB_SLAM.dir/build.make:438: recipe for target 'CMakeFiles/ORB_SLAM.dir/src/ORBextractor.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM.dir/src/ORBextractor.cc.o] Error 1
CMakeFiles/Makefile2:227: recipe for target 'CMakeFiles/ORB_SLAM.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Thanks in advance so much!

PleasantCoding, 2018-11-29:
I met this problem too.

PleasantCoding, 2018-11-30:
i finally found the solution in https://github.com/raulmur/ORB_SLAM/issues/44