raulmur/ORB_SLAM #151: Problem running ORB_SLAM (rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.yml Data/Settings.yaml)

latifAnjum, 2017-08-23:
My launch command:
rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.yml Data/Settings.yaml
Generates this error.
image
Files (ORBvoc.yml and Settings.yaml) are placed in the mentioned path, unzipped.

System Info:
I have Ubuntu 16.04 with ROS Kinetic Kame. Please help. I have successfully built ORBSLAM in ROS Kinetic Kame, so I guess this distro of ROS is not the problem.

Gasethata, 2017-09-13:
I think I have a similar issue as yours.
I used the ORBvoc.txt then edited the TUM1.yaml (only changed the camera calibration parameters) and use it as my SETTINGS_FILE, and the following is what I get, till I terminate.

vip@vip-OptiPlex-7040:~/ORB_SLAM2$ rosrun ORB_SLAM2 Mono /home/vip/ORB_SLAM2/Vocabulary/ORBvoc.txt /home/vip/ORB_SLAM2/Examples/ROS/ORB_SLAM2/SettingsFile/microsoftWebcam.yaml
[ERROR] [1505333162.065357875]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

Then this:

Input sensor was set to: Monocular
Failed to open settings file at: /home/vip/ORB_SLAM2/Examples/ROS/ORB_SLAM2/SettingsFile/microsoftWebcam.yaml

Can someone please help.

wangzhaoguang, 2017-09-14:
I met the similar issues.

[image]

Gasethata, 2017-09-14:
I solved the problem "Input file is empty in function cvOpenFileStorage"

The cvOpenFileStorage function reads .XML or .YAML of your camera that you have to have in the Settings_File.
I replaced the camera calibration parameters of TUMX.YAML with my camera calibration file i got after opencv calibration.

The camera calibration file was not be edited, I just copied it and replaced those in the TUMX.YAML.
Hope this helps.

wangzhaoguang, 2017-09-14:
@Gasethata hello, you mean you edited Settings.yaml file?

latifAnjum, 2017-09-18:
@Gasethata: Isn't your code related to ORB_SLAM2? That is different than ORB_SLAM.

Gasethata, 2017-09-20:
@wangzhaoguang I edited the TUM1.YAMAL file from the ~/ORB_SLAM2/Examples/Monocular, where I replaced the calibration parameters after I calibrated my camera (Microsoft Webcam Model:1407)....and also replaced the frame rate (fps --> Frames Per Seconds) with that of my camera.

Gasethata, 2017-09-20:
@latifAnjum I'm using ORB_SLAM2, I haven't played with ORB_SLAM yet.
So I'm not sure if they are different or not....but I guess if they are different, it shouldn't be that much.

johnchars, 2019-04-13:
I met same error, but fixed the path it worked.
```
$ pwd
~/ORB_SLAM
rosrun ORB_SLAM ORB_SLAM Data/ORBvoc.txt Data/Settings.yaml
```
May help you!