raulmur/ORB_SLAM #157: Re-localize a single frame?

Seanmatthews, 2018-03-14:
Is it possible to re-localize a single image frame, given a map that was previously constructed by ORB-SLAM? Is there an easy way to accomplish this with the given source code?

laxnpander, 2018-03-18:
What do you mean? Relocalization is - as far as I know - always a single frame operation. You extract features, compare them to your data base features and see if they get recognized. Problems occure, if your features are slightly different -> bad lighting / bad rotation angle / bad something. I am not sure, if there is really a universal solution to this problem that ALWAYS works. And if so, it might need a lot of time

Mechazo11, 2023-02-08:
Are you referring to the problem of figuring out where in the physical space the given frame is most similar to or recovering from a short-term loss?