raulmur/ORB_SLAM #167: final problem in build when running make command.

PleasantCoding, 2018-11-30:
`lee@lee-Aspire-VN7-592G:/media/lee/ubuntu/ORB_SLAM/build$  cmake .. -DROS_BUILD_TYPE=Release
[rosbuild] Building package ORB_SLAM
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
-- Using CATKIN_DEVEL_PREFIX: /media/lee/ubuntu/ORB_SLAM/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /media/lee/ubuntu/ORB_SLAM/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[rosbuild] Including /opt/ros/kinetic/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/kinetic/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/kinetic/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Configuring done
-- Generating done
-- Build files have been written to: /media/lee/ubuntu/ORB_SLAM/build
lee@lee-Aspire-VN7-592G:/media/lee/ubuntu/ORB_SLAM/build$ make
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
Scanning dependencies of target ORB_SLAM
[  5%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/main.cc.o
[ 10%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Tracking.cc.o
[ 15%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/LocalMapping.cc.o
[ 21%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/LoopClosing.cc.o
[ 26%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/FramePublisher.cc.o
[ 31%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Converter.cc.o
[ 36%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/KeyFrame.cc.o
[ 42%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Optimizer.cc.o
[ 47%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Frame.cc.o
[ 52%] Building CXX object CMakeFiles/ORB_SLAM.dir/src/Initializer.cc.o
[ 57%] Linking CXX executable ../bin/ORB_SLAM
/usr/bin/ld: CMakeFiles/ORB_SLAM.dir/src/main.cc.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
/usr/lib/x86_64-linux-gnu/libboost_system.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/ORB_SLAM.dir/build.make:1609: recipe for target '../bin/ORB_SLAM' failed
make[2]: *** [../bin/ORB_SLAM] Error 1
CMakeFiles/Makefile2:324: recipe for target 'CMakeFiles/ORB_SLAM.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`

PleasantCoding, 2018-11-30:
By the way, my system version is ubuntu16.04 and ros version is ROS kenetic. I have no idea of this problem.

hca2014, 2018-11-30:
You need to include boost-system in your CMakeLists.txt file. There is a similar problem reported for ORBSLAM2 #552. Also there seems an easier way in an earlier ticket.Do a search there. -lboost-system to be added.

PleasantCoding, 2018-12-02:
> You need to include boost-system in your CMakeLists.txt file. There is a similar problem reported for ORBSLAM2 #552. Also there seems an easier way in an earlier ticket.Do a search there. -lboost-system to be added.

thank you very much ,I finally solve this problem!
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
[  5%] Linking CXX executable ../bin/ORB_SLAM
[100%] Built target ORB_SLAM

sumitsarkar1, 2019-09-26:
How did you solve?

Tamire, 2020-08-27:
How did you solve this problem? I am stuck at the same point.
getting:

> /usr/bin/ld: CMakeFiles/ORB_SLAM.dir/src/main.cc.o: undefined reference to symbol '_ZN5boost6system15system_categoryEv'
/usr/lib/x86_64-linux-gnu/libboost_system.so: error adding symbols: DSO missing from command line

I tried adding to CMakeLists.txt:
set(LIBS
-lboost_system)

didn't help :(

wanjh1024, 2021-03-06:
add to CMakeLists.txt:
target_link_libraries(${PROJECT_NAME}
 ${OpenCV_LIBS}
 ${EIGEN3_LIBS}
 ${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so
 ${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib/libg2o.so
 /usr/lib/x86_64-linux-gnu/libboost_system.so
 /usr/lib/x86_64-linux-gnu/libboost_filesystem.so
 )