raulmur/ORB_SLAM #170: CMake Error

romi-seecs, 2018-12-25:
I getting this error, can anyone help me out here?

```
romi@ubuntu:~/ORB_SLAM/build$ cmake ..
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
[rosbuild] Building package ORB_SLAM
[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/romi/ORB_SLAM
1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM" at "", but the current
  directory is "/home/romi/ORB_SLAM".  You should double-check your
  ROS_PACKAGE_PATH to ensure that packages are found in the correct
  precedence order.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/romi/ORB_SLAM/build/CMakeFiles/CMakeOutput.log".

```

RahmanMifta, 2018-12-25:
I think you have to source the ROS workspace of ORB-SLAM.

romi-seecs, 2018-12-25:
YUP! It worked after  export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/romi

closed 2018-12-25 by romi-seecs